PaulStoffregen/CapacitiveSensor #6: Exception 29 on ESP8266-12e

s00500, 2015-10-05:
Hi I'm getting an Exception (29) on the esp8266-12E (NodeMCU DevBoard v1.0 "Amica")
Here is the error:

```
Exception (29):
epc1=0x40201e82 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000008 depc=0x00000000

ctx: cont 
sp: 3ffe9b60 end: 3ffe9d30 offset: 01a0

>>>stack>>>
3ffe9d00:  3fffdc20 000000fa 3ffe9d54 40201cf4  
3ffe9d10:  3fffdc20 00000000 3ffe9d54 4020184a  
3ffe9d20:  00000000 00000000 3ffe8d10 40100118  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld
```

PaulStoffregen, 2015-10-05:
I can't troubleshoot ESP8266 problems.  But if someone does solve this, please submit the fix as a pull request.  As long as it only changes the ESP8266-specific code, I'll be happy to merge the fix.

develpr, 2015-11-04:
@s00500 if you don't mind me asking, how did you have your circuit setup for this? Are you using the ADC pin?

s00500, 2015-11-05:
No, I havn't  been using D5 and D6 (GPIO 12 and 14)

androcioli, 2016-02-17:
@s00500 Hi I'm getting an Exception (29) on the esp8266 too. Did you resolve this problem with CapacitiveSensor or you have found another code for the ESP8266?

s00500, 2016-02-17:
@androcioli no i've been using way simpler code to do this.

androcioli, 2016-02-17:
@s00500 Do you can making the code available you're using?

BorisWild, 2016-02-21:
Open CapacitiveSensor.h... find lines 95 - 104... and refresh them with that: 
#elif defined(ARDUINO_ARCH_ESP8266)
#define PIN_TO_BASEREG(pin)             ((volatile uint32_t*) GPO)
#define PIN_TO_BITMASK(pin)             (1 << pin)
#define IO_REG_TYPE uint32_t
#define IO_REG_ASM
#define DIRECT_READ(base, mask)         ((GPI & (mask)) ? 1 : 0)    //GPIO_IN_ADDRESS
#define DIRECT_MODE_INPUT(base, mask)   (GPE &= ~(mask))            //GPIO_ENABLE_W1TC_ADDRESS
#define DIRECT_MODE_OUTPUT(base, mask)  (GPE |= (mask))             //GPIO_ENABLE_W1TS_ADDRESS
#define DIRECT_WRITE_LOW(base, mask)    (GPOC = (mask))             //GPIO_OUT_W1TC_ADDRESS
#define DIRECT_WRITE_HIGH(base, mask)   (GPOS = (mask))             //GPIO_OUT_W1TS_ADDRESS

PaulStoffregen, 2016-02-21:
fixed by https://github.com/PaulStoffregen/CapacitiveSensor/commit/83ee030b7c20a5eb7f127f900cfcc95f886642e2

closed 2016-02-21 by PaulStoffregen

Maxr1998, 2016-11-02:
@BorisWild I somehow can't get it to work for me, it always throws a timeout (-2). This is my code:
```
#include <CapacitiveSensor.h>

CapacitiveSensor csense = CapacitiveSensor(12, 14);

void setup() { 
  Serial.begin(115200);
}

void loop() {
  Serial.println(csense.capacitiveSensor(30));
}
```

I have a 10MOhm resistor wired up to pin 12, and a 1KOhm resistor to pin 14. They are soldered together on a 1cm*3cm sheet of copper tape. Do I have to search the error in the code or in the wiring?

BorisWild, 2016-11-03:
Too big resistors.

Maxr1998, 2016-11-03:
Well, example code said:
> Uses a high value resistor e.g. 10M between send pin and receive pin
 > Resistor effects sensitivity, experiment with values, 50K - 50M. Larger resistor values yield larger sensor values.
 > Receive pin is the sensor pin - try different amounts of foil/metal on this pin

What values would you recommend for the ESP then?

Maxr1998, 2016-11-03:
Ok, I tested my resistor values on an Arduino Mega, where they work fine.
Now I just need to adapt them to my ESp8266.

rmeister, 2016-11-22:
Came here for the same problem as Maxr1998, but found out that in my case the error reason was another: The library does not support GPIO16 on the ESP8266.

I use a 10K resistor for direct touch sensing.

Maxr1998, 2016-11-26:
Forgot to update, it also works for me now, with a 100k to 200k resistor through a 1mm layer of plastic.