PaulStoffregen/CapacitiveSensor #15: Wemos D1 mini (esp-8266) - Error -2

batizisanya, 2017-01-30:
In serial monitor i get -2, but when i touch both input and receive pin i get normal value (0-40000). What can i do?

jonnor, 2017-01-30:
`-2` indicates that the senso reading timed out. If you just want a binary switch, then treat this as "no one is touching". If you want to try to sense analog values better, you can increase the timeout value using `set_CS_Timeout_Millis()`.

jonnor, 2017-01-30:
Looking at the timeout handling code, I see a hardcoded division by `16000000` (16Mhz) though, so it is likely that on non-AVR devices, like the ESP8266, that the values are off by quite a bit - because the CPU clock frequency is much higher.

batizisanya, 2017-01-30:
Few minutes ago, i tried with ac/dc USB adapter, not with laptop, And it worked. I don't know why don't work whit laptop, When its Powered to AC.

batizisanya, 2017-01-31:
Can you send an example code what tested on wemos (esp-8266), please? I tried with 1, 10, 20, 50 Kohm resistors, but when i tried thruw glass, noone switched the led. (with the default example sketch if total1 > 0 {digitalWrite(led, LOW)};)

Its just change when i touch the foil, but when i touch glass (under foil) its not work.

P.s.: sorry for my bad english

PaulStoffregen, 2017-01-31:
I do not use ESP chips, so I depend on the ESP community to test this stuff and submit good pull requests when something needs to be updated for ESP.

This issue tracker is **NOT** the place to ask for technical assistance with your project.  It's meant only for reporting problems and contributing fixes to this software.

Obviously I can't help with resolving any ESP problem.

closed 2017-01-31 by batizisanya