PaulStoffregen/CapacitiveSensor #16: Analog pin affecting capacitive sensor values

antonbilbaeno, 2017-04-10:
I am reading the analog value from a pot connected to VCC and GND with the wiper connected to A0.

Turning this pot varies my capacitive sensor (send pin 3, rcv pin 2) by a constant value which increases as the voltage at A0 increases.

Just wondering if this has ever been seen before.
Any instincts as to why?

Maybe it has to do with:
-slight voltage variance on VCC (or, AREF pin?) at different pot positions
--maybe I can recalibrate after detecting that the pot position has change?
--how is the recalibration function implemented?
-I am also outputting a 12-32kHz PWM signal proportional to the analog value that I read.

PaulStoffregen, 2017-04-10:
Have seen PWM and other waveforms like class D audio amps mess up capacitive sensing.  The large number of rapid digital switching can easily couple to whatever you're sensing.  Those wires need to be kept as far apart as possible.

antonbilbaeno, 2017-04-10:
I believe the problem is that I'm messing up the timer which is used for millis() since you use that.

I thought millis() was on Timer0.
I thought I was avoiding this by using Timer1?

This is the line that sets the counter for Timer1.
OCR1A = (C_RANGE * c_norm) + C_MIN;

Maybe I have a fundamental mis-understanding around the Timers.

antonbilbaeno, 2017-04-11:
I don't know enough to explain why, but seem to have worked around the issue by using the mode where the pin is toggled based on the OCRA (output compare register). I was previously using the ISR to toggle a pin that I defined, but now I am using the pin that is tied directly in hardware.

Can't explain why this has such a dramatic effect.

PaulStoffregen, 2017-04-11:
Github issues are the wrong place for this sort of project help.  This really needs to be discussed on the forums.

closed 2017-04-11 by PaulStoffregen