PaulStoffregen/CapacitiveSensor #18: Error -2 when using the ESP8266

digitalLumberjack, 2017-04-30:
Hi, I just received a NodeMcu ESP8266-12 and I discovered your lib, thank you for the good work.

There are something that bother me though: 
- when I run the example code, with a simple 1Mohm resistor, when nobody touches the foil, the lib returns -2 when calling `.capacitiveSensor(30);`
- I touch the foil, the sensor behave normally and return
So i tried with 3Mohm to see if it changes anything, but still the same.

I tested with 3 different power supply, but that changed nothing.

The solution is to put a 300kohm resistor. It works really well, return 0 to 2 when nothing happen, to 5000/6000 when someone is touching the foil.

Maybe the documentation can be updated with those values.

Have fun and thanks for the lib :)

Bimblis, 2017-07-27:
Is it there anyway of making it work from a distance, instead of just "by touch", using the ESP8266/NodeMCU?

Using different resistors, but all still seem to work just "by touch".

jtroberts1, 2017-10-27:
@digitalLumberjack  would you mind posting your code.  I can get this to work on an Arduino uno but not on my Wemos D1 mini.  Any help is appreciated.  Thanks

digitalLumberjack, 2017-10-27:
https://gitlab.com/iot4i/firmware/blob/master/src/main.cpp 
ony work "by touch"

javiercuellar73, 2018-10-30:
Same issue here. Would be nice to make it as sentitive as when used with UNO.
I think it has something to do because internal pull-up/down resistors that comes with ESP8266.

> Hi, I just received a NodeMcu ESP8266-12 and I discovered your lib, thank you for the good work.
>
> There are something that bother me though:
>
> * when I run the example code, with a simple 1Mohm [...]