PaulStoffregen/CapacitiveSensor #19: Always timeout (-2) with Wemos D1 Mini and 1MOhm resistor

trancefreak77, 2017-05-11:
I can't get a sensor reading when I use a 1MOhm resistor or above. Only < 600kOhm work but then I can only sense an absolute touch but I need to get a signal when someone is about 2 cm away from the touch sensor field.

The Wemos D1 Mini has 3.3V pins.

Does anyone know why resistors > 600k always time out? Lower resistor values work but will only detect absolute touches :-/

tradeJmark, 2023-01-31:
I'm having a similar problem on an ESP32, only I can't even use anything above about 50k.

tradeJmark, 2023-01-31:
Seems the resistors are dropping the voltage too much for the read pin to recognize the value going high (I think). Haven't worked out a solution yet.