PaulStoffregen/CapacitiveSensor #24: does it support ESP32 board ?

adrianmak, 2018-01-10:
I owned this board
https://www.banggood.com/ESP32-Development-Board-WiFiBluetooth-Ultra-Low-Power-Consumption-Dual-Cores-ESP-32-ESP-32S-Board-p-1109512.html

I tried to copy from demo code for testing but firmware ha built error.

`#include "CapacitiveSensor.h"

CapacitiveSensor cs_4_2 = CapacitiveSensor(4, 2);

void setup() {
    cs_4_2.set_CS_AutocaL_Millis(0xFFFFFFFF);
    Serial.begin(9600);

}

void loop() {

    long start = millis();
    long total1 = cs_4_2.capacitiveSensor(30);

    Serial.print(millis() - start);
    Serial.print("\t");
    Serial.print(total1);
    Serial.print("\t");
    delay(10);
}
`

In file included from src\main.cpp:3:0:
lib\CapacitiveSensor/CapacitiveSensor.h:239:2: error: 'IO_REG_TYPE' does not name a type
IO_REG_TYPE sBit;   // send pin's ports and bitmask
^
lib\CapacitiveSensor/CapacitiveSensor.h:240:11: error: 'IO_REG_TYPE' does not name a type
volatile IO_REG_TYPE *sReg;
^
lib\CapacitiveSensor/CapacitiveSensor.h:241:2: error: 'IO_REG_TYPE' does not name a type
IO_REG_TYPE rBit;    // receive pin's ports and bitmask
^
lib\CapacitiveSensor/CapacitiveSensor.h:242:11: error: 'IO_REG_TYPE' does not name a type
volatile IO_REG_TYPE *rReg;

PaulStoffregen, 2018-01-15:
I do not use ESP.  I'm depending on the ESP community to resolve problems and submit pull requests for this open source software, to improve its support for ESP.

ba58smith, 2018-11-12:
Thanks. One more quickie - if you can. Can I assume that the Arduino IDE (when you select the Board you are using) takes care of defining the controller you're using? IOW, if I properly define the PIN_TO_BASEREG, etc. (above) for a particular board (Wemos D1 mini, for example), and then select the Wemos D1 mini in the Arduino IDE... that OUGHT to do it?

ba58smith, 2018-12-06:
@jonmacd - I haven't gotten to it. Not sure I will. I have gone off in a different direction for my project for the time being. I'm hoping that Paul's reply above ("That should be it. Look at OneWire's headers.") is all you need to do, though. I did look at the headers for OneWire, and did find a section for ESP8266, so if I were going to try to get this to work, I think I'd make the ___ elif defined ___ section look like the ESP8266 section for the OneWire header. Good luck.

jonmacd, 2018-12-06:
> @jonmacd - I haven't gotten to it. Not sure I will. I have gone off in a different direction for my project for the time being. I'm hoping that Paul's reply above ("That should be it. Look at [...]

Thanks for the reply. Yeah, sorry, I was just using the ESP8266 as an example in my comment because I figured it would be similar (it appears the CapacitiveSensor library already supports the ESP8266). I am indeed trying to use this with an ESP32.

I managed to track down the `#elif defined(ARDUINO_ARCH_ESP32)` section in the [OneWire header file](https://github.com/PaulStoffregen/OneWire/blob/master/util/OneWire_direct_gpio.h) Paul mentioned. I just pasted that entire section into the CapacitiveSensor.h file with all the other `#elif defined` sections. Everything compiles no problem. Using a 10Mohm resistor, at first I couldn't get any readings using a few combinations of touch pins 0-3 on my ESP32 (not sure what's going on there. Moved my resistor to the other side of the board using pins 27 and 12 and I've got nice looking data streaming in.

Thanks!

ba58smith, 2018-12-06:
Glad you got it working! I think you can make a pull request on this GitHub page to add what you did, for future reference. I think Paul will accept it and update it - he just isn't doing any active development.

06GitHub, 2020-11-11:
> [jonmacd] I managed to track down the `#elif defined(ARDUINO_ARCH_ESP32)` section in the [OneWire header file](https://github.com/PaulStoffregen/OneWire/blob/master/util/OneWire_direct_gpio.h) Paul mentioned. I just pasted that entire section into the CapacitiveSensor.h file with all the other `#elif defined` sections. Everything compiles no problem. Using a 10Mohm resistor, at first I couldn't get any readings using a few combinations of touch pins 0-3 on my ESP32 (not sure what's going on there). Moved my resistor to the other side of the board using pins 27 and 12 and I've got nice looking data streaming in.
> 
> Thanks!

@jonmacd Could you please share your CapacitiveSensor.h file which is ok for ESP32 ? Errors below occur when importing the CapacitiveSensor Library (#include <CapacitiveSensor.h>) and compiling for ESP32. Those errors are identical to ones described in https://github.com/PaulStoffregen/CapacitiveSensor/issues/24#issue-287302477
Thanks !

> [Nov 11, 2020]C:\Users\Admin\Documents\Arduino\libraries\CapacitiveSensor/CapacitiveSensor.h:222:2: error: 'IO_REG_TYPE' does not name a type
>   IO_REG_TYPE sBit;   // send pin's ports and bitmask
>   ^
> C:\Users\Admin\Documents\Arduino\libraries\CapacitiveSensor/CapacitiveSensor.h:223:11: error: 'IO_REG_TYPE' does not name a type
>   volatile IO_REG_TYPE *sReg;
>            ^
> C:\Users\Admin\Documents\Arduino\libraries\CapacitiveSensor/CapacitiveSensor.h:224:2: error: 'IO_REG_TYPE' does not name a type
>   IO_REG_TYPE rBit;    // receive pin's ports and bitmask
>   ^
> C:\Users\Admin\Documents\Arduino\libraries\CapacitiveSensor/CapacitiveSensor.h:225:11: error: 'IO_REG_TYPE' does not name a type
>   volatile IO_REG_TYPE *rReg;
>            ^
> Using the CapacitiveSensor version 0.5.1 library

MIKHANYA, 2021-12-10:
Same question, how to make this library work on ESP32 ? (I use esp-c3-32s) Guide described here https://github.com/PaulStoffregen/CapacitiveSensor/issues/24#issuecomment-444718037 don't work.

PaulStoffregen, 2021-12-17:
First, you must find someone from the ESP community with knowledge of using the ESP's hardware registers.

The abstraction layer this library uses is nearly the same as OneWire.  So perhaps an expert in ESP hardware registers could copy the OneWire defines and get this library to work?

If so, please send a pull request when it is confirmed working.  I do not personally use ESP, so I'm depending on the ESP community to submit pull requests.  This generic guidance to copy the nearly-identical defines from OneWire is as much as I can do to help.

arjuna-dev, 2022-09-21:
Actually the code is already updated to work properly with esp32 boards in the master branch of this repo but in the released version you can get from within the Arduino IDE there is still no support for esp32

tradeJmark, 2023-01-31:
Are you sure about that? When I try using the latest master to compile for ESP32 boards, I get `'rtc_gpio_desc' was not declared in this scope' and `'esp32_gpioMux' was not declared in this scope` and it fails to build.

tradeJmark, 2023-01-31:
Update: It will compile for ESP32... on pre-v2 versions of the ESP board manager. It doesn't work on recent versions. I think we need to do something like [this](https://github.com/PaulStoffregen/OneWire/pull/114), but I'm not sure I understand it intimately enough to do it myself.

callagga, 2023-10-29:
no updates to this I don't suppose (re fixing the 'rtc_gpio_desc' was not declared in this scope) issue