PaulStoffregen/CapacitiveSensor #37: Arduino Nano 33 BLE compatibility

zsmatyas, 2020-04-21:
### Description

Library does not compile on Arduino Nano 33 BLE

### Steps To Reproduce Problem
Download latest library. It is v0.5.1 as of today.

### Hardware & Software

Board Arduino Nano 33 BLE
Shields / modules used none
Arduino IDE version 1.8.12 
Teensyduino version (if using Teensy) none
Version info & package name (from Tools > Boards > Board Manager) v1.1.4 Arduino nRF528x
Operating system & version Windows 10
Any other software or hardware? none.

### Arduino Sketch
none.

```cpp
#include <CapacitiveSensor.h>

void setup() { 
// intentionally empty
}

void loop() {
// intentionally empty
}
```

### Errors or Incorrect Output

```
CapacitiveSensor.h:222:2: error: 'IO_REG_TYPE' does not name a type

  IO_REG_TYPE sBit;   // send pin's ports and bitmask

  ^~~~~~~~~~~

CapacitiveSensor.h:223:11: error: 'IO_REG_TYPE' does not name a type

  volatile IO_REG_TYPE *sReg;

           ^~~~~~~~~~~

CapacitiveSensor.h:224:2: error: 'IO_REG_TYPE' does not name a type

  IO_REG_TYPE rBit;    // receive pin's ports and bitmask

  ^~~~~~~~~~~

CapacitiveSensor.h:225:11: error: 'IO_REG_TYPE' does not name a type

  volatile IO_REG_TYPE *rReg;

           ^~~~~~~~~~~

exit status 1
Error compiling for board Arduino Nano 33 BLE.
```

80prozent, 2020-05-03:
same issue here. 
would really like to use this lib and my arduino nano BLE 33 to build a capacity sensor.

are there plans to support arduino nano BLE 33 ?

PaulStoffregen, 2020-05-03:
The "plan" is to wait for someone from the open source community to craft & contribute the direct I/O defines.

80prozent, 2020-05-04:
i am fairly new to all the arduino stuff.

I will look into this and if i get something to work i will share it, but cant make any promises...

paoudet, 2020-05-25:
Hello,
I was interested in this too as I had developped a project on Arduino Uno and wanted to move it to the NANO 33 BLE.
After a lot and lot of simplification I basically ended up with something which compiles and is dedicated for the Arduino. unfortunately, compiling doesn't mean it works.
My simplest version still works on the UNO. 
After many experiments and tracing I found out that the ports behaviour seem to be different on the nano, as if there were some internal pull-ups which made the code totally ineffective.
Basically to test the functionality I ended writing something like
- put sendpin to high
- read receivepin
- put sendpin to low
- read receivepin
with some delays in between. basically receive pin reads always as high (unless I giggle it with a multimeter test point). I've tried with D2 and D3 with same result.
I looked at the NINA datasheet and the pullups don't seem to be activated by default so I don't understand (and at reset the pin is measured as low). and I can tell you I have spent hours with various codes (all work on the UNO).
That's quite a disappointment as I expected the code would be portable to all the platforms. At this point I have no idea how to solve this (I have thought using an analog input but haven't tried yet)

which raises a different question: is there a way in the code to directly address the MCU registers 
(I am almost totally new to the arduino world)
thanks
Pascal

80prozent, 2020-05-26:
Hey paoudet

i got to a similar point as you did. i tried a while to get the lib to compile for the Nano BLE 33, but could not get it to work.
So i tried just to mimic what the lib is doing with my own code specially for the Nano BLE, but did not got it to really work either.
I also tested this lib: https://github.com/Nyanyan/FastCapacitiveSensor which works with analog-inputs rather than standart gpio pins, and while the results was promising using 1 analog input, i had strange results with it aswell when i started to try to read multiple analog inputs...

I kind of gave up at that point, and ordered a few of those: https://learn.adafruit.com/adafruit-mpr121-12-key-capacitive-touch-sensor-breakout-tutorial/wiring

Only got around to do some basic tests with one of them, but so far i like the chip. does what it supposed to do and gives me 12 touch-readings.

@PaulStoffregen: i will probably get back to test your lib at some point when i have a different arduino at hand...

Cheers
Robin

paoudet, 2020-06-16:
Hello

just to keep those who might encounter this issue or a similar one updated, in case it helps
After much investigation digging directly into the NRF hardware registers, I found where the problem comes from:
when accessing directly the GPIO_IN registers, the values are consistent: send 0, receive 0
However, if I use digitalRead I got 1 on receive
What I found is that a call to digitalRead changes the pin CNF register and enables the internal pullups (CNF changes from 0 (input, buffer, no pull up) to 0x0C (input, buffer, pull up)
Pascal

sbhklr, 2020-08-26:
If anyone wants to test it, here's the PR: https://github.com/PaulStoffregen/CapacitiveSensor/pull/39
I tested it with my BLE Sense and it works but I'd like to have confirmation from you too.

closed 2020-08-26 by PaulStoffregen