PaulStoffregen/CapacitiveSensor #45: Update the version and create a new release

davepruitt, 2023-01-17:
It seems like there has been a lot of commits over the past couple of years, but the most recent official release is from 2016. So if I were to download this library through the Arduino IDE, I would get code from 6 years ago. It would probably be wise to create a new official release so that anyone who downloads this through the Arduino IDE gets the most up-to-date code.

tradeJmark, 2023-01-30:
Or at least create some release tags so people using it from PlatformIO can specify the dependency without having to pull directly from master or put a commit ID into the dependency string. It's even easier than rolling a release zip, and a good intermediary step even if the mainainer doesn't have time to publish a release.

gmglickman, 2023-04-30:
And that's the reason it won't compile for ESP8266 if installed via Library Manager; need to download zip file and install.