PaulStoffregen/CapacitiveSensor #48: Build error on Aduino GIGA STM32

LeFab64, 2024-01-19:
Hello,
I'm using the last version of this library on an Arduino GIGA but I have error during building the example code.

error: 'IO_REG_TYPE' does not name a type IO_REG_TYPE sBit;   // send pin's ports and bitmask

Is this library compatible with STM32 processors?

Regards.

ozatyx, 2024-02-15:
Hey LeFab,
I managed to fix the same issue on my ESP32C3 by installing the OneWire library and changing some lines of code.

In CapacitiveSensor.h from lines 33-37, I replaced with the following code:

```
#ifndef CapacitiveSensor_h
#define CapacitiveSensor_h
#include "util/OneWire_direct_gpio.h"
#include "Arduino.h"
```

I believe that #include util/OneWire_direct_gpio.h is the only thing you need to add since you are on an Arduino platform. I got rid of the stuff with WProgram.h since it doesn't exist on ESP32.

If you downloaded the library through Platformio on VScode, some code is outdated in CapacitiveSensor.cpp. Replace line 77 with this:

```
unsigned long diff = (total > leastTotal) ? total - leastTotal : leastTotal - total;
		if ( (millis() - lastCal > CS_AutocaL_Millis) && diff < (int)(.10 * (float)leastTotal) ){
```

Hope this helps!

LeFab64, 2024-02-19:
Hi Ozatyx,
Thanks a lot for your answer I'll try this solution asap...
Regards.